shattered97/CameraTracker #5: Setup Folder Structure on Linux - do this prior to setting up QT

shattered97, 2020-01-17:
We want remove permission issues so this project will run as root.  

`cd /`
`sudo mkdir checkouts/`
`sudo chmod 775 /checkouts`
Confirm you permissions look like this:
` ls -l`
total 2097264
drwxr-xr-x   2 root root       4096 Jan 17 15:22 bin
drwxr-xr-x   3 root root       4096 Jan 17 15:44 boot
drwxrwxr-x   2 root root       4096 Jan 17 13:43 cdrom
drwxrwxr-x   2 root root       4096 Jan 17 16:26 checkouts
drwxr-xr-x  18 root root       4120 Jan 17 15:47 dev
drwxr-xr-x 132 root root      12288 Jan 17 15:53 etc
drwxr-xr-x   3 root root       4096 Jan 17 13:44 home
lrwxrwxrwx   1 root root         32 Jan 17 15:25 initrd.img -> boot/initrd.img-5.0.0-38-generic
lrwxrwxrwx   1 root root         33 Jan 17 15:25 initrd.img.old -> boot/initrd.img-4.15.0-74-generic
drwxr-xr-x  21 root root       4096 Jan 17 15:52 lib
drwxr-xr-x   2 root root       4096 Jan 17 15:52 lib32
drwxr-xr-x   2 root root       4096 Jan 17 15:07 lib64
drwx------   2 root root      16384 Jan 17 13:38 lost+found
drwxr-xr-x   3 root root       4096 Jan 17 14:10 media
drwxr-xr-x   2 root root       4096 Jul 24  2018 mnt
drwxr-xr-x   2 root root       4096 Jul 24  2018 opt
dr-xr-xr-x 201 root root          0 Jan 17 15:47 proc
drwx------   4 root root       4096 Jan 17 14:56 root
drwxr-xr-x  30 root root        860 Jan 17 15:49 run
drwxr-xr-x   2 root root      12288 Jan 17 15:43 sbin
drwxr-xr-x  12 root root       4096 Jan 17 15:35 snap
drwxr-xr-x   2 root root       4096 Jul 24  2018 srv
-rw-------   1 root root 2147483648 Jan 17 13:38 swapfile
dr-xr-xr-x  13 root root          0 Jan 17 15:47 sys
drwxrwxrwt  20 root root       4096 Jan 17 16:26 tmp
drwxr-xr-x  12 root root       4096 Jan 17 15:52 usr
drwxr-xr-x  14 root root       4096 Jul 24  2018 var
lrwxrwxrwx   1 root root         29 Jan 17 15:25 vmlinuz -> boot/vmlinuz-5.0.0-38-generic
lrwxrwxrwx   1 root root         30 Jan 17 15:25 vmlinuz.old -> boot/vmlinuz-4.15.0-74-generic

Jetstream-dev, 2020-01-18:
Errors running the 3rd line of code. 
The chmod, chown, and chgrp return that an operand is needed after the checkouts.
Should the '\' be '/' ?

shattered97, 2020-01-18:
Each of those is one line or one command, I fixed it so it was more clear.

chmod - changes the permissions of the folder
chown - changes the owner of the folder
chgrp - changes the group of the folder

if you need help with linux commands you can type man <command>
`$man chmod'

Jetstream-dev, 2020-01-18:
Is this the expected result?

sudo chmod 775 /checkouts - returns nothing
sudo chown \checkouts - missing operand after 'checkouts'
sudo chgrp \checkouts - missing operand after 'checkouts'
ls -l
total 2097260
drwxr-xr-x   2 root root       4096 Jan 17 20:07 bin
drwxr-xr-x   4 root root       4096 Jan 17 20:13 boot
drwxrwxr-x   2 root root       4096 Jan  3 18:07 cdrom
drwxrwxr-x   2 root root       4096 Jan 17 20:17 checkouts
drwxr-xr-x  20 root root       4460 Jan 17 21:22 dev
drwxr-xr-x 137 root root      12288 Jan 17 21:21 etc
drwxr-xr-x   3 root root       4096 Jan  3 18:08 home
lrwxrwxrwx   1 root root         32 Jan 17 20:07 initrd.img -> boot/initrd.img-5.0.0-38-generic
lrwxrwxrwx   1 root root         32 Jan 17 20:07 initrd.img.old -> boot/initrd.img-5.3.0-26-generic
drwxr-xr-x  21 root root       4096 Jan 17 20:03 lib
drwxr-xr-x   2 root root       4096 Jan 17 20:01 lib64
drwx------   2 root root      16384 Jan  3 18:06 lost+found
drwxr-xr-x   3 root root       4096 Jan  3 18:14 media
drwxr-xr-x   2 root root       4096 Aug  5 13:58 mnt
drwxr-xr-x   4 root root       4096 Jan 14 19:16 opt
dr-xr-xr-x 247 root root          0 Jan 17 21:22 proc
drwx------   6 root root       4096 Jan 17 19:56 root
drwxr-xr-x  31 root root        880 Jan 17 21:23 run
drwxr-xr-x   2 root root      12288 Jan 17 21:21 sbin
drwxr-xr-x  11 root root       4096 Jan  3 18:15 snap
drwxr-xr-x   2 root root       4096 Aug  5 13:58 srv
-rw-------   1 root root 2147483648 Jan  3 18:06 swapfile
dr-xr-xr-x  13 root root          0 Jan 17 21:22 sys
drwxrwxrwt  18 root root       4096 Jan 17 21:29 tmp
drwxr-xr-x  11 root root       4096 Aug  5 14:03 usr
drwxr-xr-x  14 root root       4096 Aug  5 14:11 var
lrwxrwxrwx   1 root root         29 Jan 17 20:07 vmlinuz -> boot/vmlinuz-5.0.0-38-generic
lrwxrwxrwx   1 root root         29 Jan 17 20:07 vmlinuz.old -> boot/vmlinuz-5.3.0-26-generic

shattered97, 2020-01-18:
Close but quite...

What is your user name - is it root

change shattered to your username

chmod is good

sudo chown shattered \checkouts
sudo chgrp shattered \checkouts

Jetstream-dev, 2020-01-18:
@shattered97 if I'm wrong, move this back to in-progress
Assuming that the chown and chgrp should respond the same as chmod did, I am calling this complete.
total 2097260
drwxr-xr-x   2 root  root        4096 Jan 17 20:07 bin
drwxr-xr-x   4 root  root        4096 Jan 18 14:08 boot
drwxrwxr-x   2 root  root        4096 Jan  3 18:07 cdrom
drwxrwxr-x   2 tyler tyler       4096 Jan 17 20:17 checkouts
drwxr-xr-x  20 root  root        4460 Jan 18 14:07 dev
drwxr-xr-x 137 root  root       12288 Jan 17 21:21 etc
drwxr-xr-x   3 root  root        4096 Jan  3 18:08 home
lrwxrwxrwx   1 root  root          32 Jan 17 20:07 initrd.img -> boot/initrd.img-5.0.0-38-generic
lrwxrwxrwx   1 root  root          32 Jan 17 20:07 initrd.img.old -> boot/initrd.img-5.3.0-26-generic
drwxr-xr-x  21 root  root        4096 Jan 17 20:03 lib
drwxr-xr-x   2 root  root        4096 Jan 17 20:01 lib64
drwx------   2 root  root       16384 Jan  3 18:06 lost+found
drwxr-xr-x   3 root  root        4096 Jan  3 18:14 media
drwxr-xr-x   2 root  root        4096 Aug  5 13:58 mnt
drwxr-xr-x   4 root  root        4096 Jan 14 19:16 opt
dr-xr-xr-x 280 root  root           0 Jan 18 14:07 proc
drwx------   6 root  root        4096 Jan 17 19:56 root
drwxr-xr-x  31 root  root         920 Jan 18 14:10 run
drwxr-xr-x   2 root  root       12288 Jan 17 21:21 sbin
drwxr-xr-x  11 root  root        4096 Jan  3 18:15 snap
drwxr-xr-x   2 root  root        4096 Aug  5 13:58 srv
-rw-------   1 root  root  2147483648 Jan  3 18:06 swapfile
dr-xr-xr-x  13 root  root           0 Jan 18 14:07 sys
drwxrwxrwt  18 root  root        4096 Jan 18 14:11 tmp
drwxr-xr-x  11 root  root        4096 Aug  5 14:03 usr
drwxr-xr-x  14 root  root        4096 Aug  5 14:11 var
lrwxrwxrwx   1 root  root          29 Jan 17 20:07 vmlinuz -> boot/vmlinuz-5.0.0-38-generic
lrwxrwxrwx   1 root  root          29 Jan 17 20:07 vmlinuz.old -> boot/vmlinuz-5.3.0-26-generic

shattered97, 2020-01-18:
@Jetstream-dev 
It is good

closed 2020-01-25 by shattered97