shattered97/CameraTracker #16: Traffic Light for a four intersection.

shattered97, 2020-01-25:
To get you started:  https://www.thegeekpub.com/16349/arduino-traffic-light-project/

Requirements:
1.  Create a new folder in the GitHub root directory to store this project.  Commit your code, when done create a push request, to create a code review.  Comment your Code with intent, descripitive variable names.
2.  When one direction is green, the other direction can not be green or yellow.
3.  Ensure both directions are red for atleast 1 sec before one direction is green.
4.  Yellow light - 3 seconds to clear the intersection.
5.  Upload video of the breadboard of the traffic light working
6.  Draw a schematic of the circuit.  Learn Eagle CAD https://www.autodesk.com/products/eagle/overview
Will integrate with fusion 360.  If Christina is in school use here school email to get a student account for fusion 360.  Eagle is free.

Questions to answer:
1.  What is the max current output per pin on the Arduino Uno?
2.  What is the current draw of each LED?
3.  What is the max current output of the Arduino Uno (All pins)?
4.  Are the six LEDs you using exceeding this?

shattered97, 2020-01-25:
All this is leading to you to the stepper motor.  You will have to understand power requirements to accomplish the stepper motor task.  The stepper motor will need an H-Bridge to control it, and it will exceed the current limits of the Arduino so you will have to design a power circuit to power the arduino and the stepper motor.  

To make things harder the micro controller board we will be using is a newer version of the AT328p you will working on.  The board is also a 3.3V logic so we have to step up and down the when communicating between the stepper and micro controller.

JetstreamDev, 2020-02-06:
@shattered97 
What is the max current output per pin on the Arduino Uno?
52mA for Elegoo Uno R3
What is the current draw of each LED?
16mA
What is the max current output of the Arduino Uno (All pins)?
500mA for Elegoo Uno R3
Are the six LEDs you using exceeding this?
No
Voltage level are the Arduino Uno pins communicating at?
5V for digital Elegoo Uno R3 pins