shattered97/CameraTracker #19: UltraSonic Radar system.

shattered97, 2020-01-25:
Using your knowledge of the servo, LEDs and ultra sonic sensor.  Build a sudo radar system with the following requirements.

1.  Arc 180 degrees.
2.  Scan everything in the 180 degree arc and build a map of all static objects.  All this means is too scan the area in front of the senor.  Save all the returned distances for each angle.  This is your static map.  
3.  Add three LEDs, one red, one green and one yellow.  When the program starts turn on the yellow led only.  After the initial scan turn on the green light.  Keep the green light on until something moves then turn the green led off, and the red led on.  The red light stays on until a user pushes a button to acknowledge the movement.  This will turn off the red light and the green light back on.  
4.  Draw a schematic of the circuit.  
5.  Create a folder on GitHub to save the code, create a pull request when all done.  Create feature branch and upload the feature branch so I can see your code, because this will be a hard project due to timing issues.  You most likely will need help.

shattered97, 2020-01-25:
The requirements are vague for a reason.

The stepper motor and the servo will be coded similar but the servo is easier to work with.

JetstreamDev, 2020-02-09:
@shattered97 
Do you want me to stop at the first difference detected or at the end of a scan if any difference is detected?
Is there a certain length of time that you want me to wait on the debounce?
Is there a set number of data points that you would like to see for each pass?

JetstreamDev, 2020-02-09:
@shattered97 
Once the button is pushed, should the new distance be added to the map?

shattered97, 2020-02-09:
@JetstreamDev 
Order should go as follows:

Start at position 1:  Read the distance in front of the sensor.  Save that distance in the map as position 1.
Move to position 2:  Read the distance in front of the sensor.  Save that distance in the map as position 2.  
Repeat this until you have covered 180 degrees.  

This is your base map.  You have read the field in front of you and stored everything in a map.  

Now you move into a search mode.  
You start at position one, read the distance and compare to your base map.  
If equal nothing moved go to the next the position
If not equal update the new position in the start map turn on a light because something moved.  

Wait until the user acknowledges the move, then continue to the next position

The goal here is not to worry about time, just the concept of how to track an object(s) that is infront of your sensor.

shattered97, 2020-02-09:
________________________________________________________________________________________________________
                                                                                                           x
                x
                                                          x

                                    servo with ultrasonic sensor

shattered97, 2020-02-09:
@JetstreamDev 
The debounce is for you figure out.  
https://www.arduino.cc/en/Tutorial/Debounce

shattered97, 2020-02-09:
@JetstreamDev 
You are building a static map.  As for when and how to do specific you as the developer get to choose how to do this.  You will be given requirements and how you get this is up to you.  

You do not have to follow any of my suggestions you have a better way.  The requirements are
Build a static map of the things in front of you.
Check if something has moved in front of you by comparing it too your static map.       
You can wait for user input before going on.
When the LED light is off the static map should reflect the current distance of any objects in front of it.
Ignore time... Do not try to do this in real time just work on looking in front of your sensor and saving those object positions, then scan again to see if anything moved. 

This is what we are going to do with the Radio beacon.  

Camera will start at a starting positioning.  
The left radio receiver will read the signal strength of the beacon.
The right radio receiver will read the signal strength of the beacon.
The goal will to be get the two signal strengths equal (or close).  This means beacon is centered between the two receivers.

There are several ways of doing this, and my testing will determine the best way.